olentangyfrc/InfiniteRecharge-2020 #7: Build electrical components for mec sled 3

olentangyfrcowner, 2020-01-08:
Need to assign others to this as well, but just getting things off my list into Git right now.

NateRaccoon26, 2020-01-11:
Took elevator off of old Proto, only mech sled we have, Tomorrow will be making a electrical board and getting it online, this will be a good way to teach about the electronics of the robots

RockyRaccoon26, 2020-01-12:
Finished Demo, Tomorrow we should finish adding the CIM motors.
I talked with Schmidt and he said we might skip the actual electronics and use the sled for prototyping only. this is to leave room for the intake and shooter systems, as the electronics sometimes get in the way

RockyRaccoon26, 2020-01-21:
Ready to be Closed

mrbbking, 2020-01-25:
I'm told this is no longer a priority, @olentangyfrcowner - can you confirm so we can close it?

closed 2020-02-21 by mrbbking