olentangyfrc/InfiniteRecharge-2020 #12: Get Color Sensors working with RoboRIO

NateRaccoon26, 2020-01-11:
We got 2 types of color sensors in, however one uses the L2C port and the other is meant for an Arduino. We need to get both working and determine which kind we will be using on the bot.

NateRaccoon26, 2020-01-11:
If I had the ability or if someone who sees this has the ability, id like to assign @mrbbking @sydneylewis22 @NateRaccoon26

mrbbking, 2020-01-11:
Thanks for adding this @NateRaccoon26 - I should be able to make assignments soon and will do this as soon as I can.

RockyRaccoon26, 2020-01-11:
https://www.mouser.com/catalog/specsheets/TCS3200-E11.pdf

RockyRaccoon26, 2020-01-12:
Just learned I won't be there tomorrow due to a hockey game, @mikagika and @worthp The Arduino and Documentation is in Schmidt's office.

mikagika, 2020-01-12:
Thanks. And I may have found a way to read it from the RIO:
[](https://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/Counter.html)

mrbbking, 2020-01-25:
Can we close this, @mikagika ? or transfer it to the Code team?

mikagika, 2020-01-25:
Yeah this is in our court. I don't know if it's going to be necessary--the Rev color sensors at first glance seemed to work ok, but it's something we can give a couple of kids a chance to poke at tomorrow.

closed 2020-01-25 by mrbbking