olentangyfrc/InfiniteRecharge-2020 #13: Create new football type boards so code team can test

NateRaccoon26, 2020-01-11:
We need to make 2 or 3 more testing boards that have at least a RoboRIO, PCM, VRM, and whatever other things code desires, Mr. King can help us with this.

NateRaccoon26, 2020-01-11:
If I had the ability or if someone sees this that has the ability, I'd like to assign, @mrbbking @sydneylewis22 @AmilSingh @HemTurakhia @NateRaccoon26

worthp, 2020-01-18:
Let's figure out how best to build out portables. Schmidt is ordering another Rio which we should have next weekend.

I think we will want the Football to have to CAN Talon controllers. I think it currently has room for 2 motors and contollers.

I think we will want to 2nd set up to have 2 SparkMax and 2 TalonSRX controllers. Not sure if we just want to connect 775s to the Talons and Neos to the Sparks. This will give us the ability to code for known hardware configs When we get the new Rio, we will prob want this same set up.

Thoughts?

mrbbking, 2020-01-25:
This is a priority for this weekend, @RockyRaccoon26 @sydneylewis22 - let's look at what Alex started and see what we need to do to finish it up.

mrbbking, 2020-01-31:
This is done (1 board created) and handed off to Code at a previous meeting. Closing.

closed 2020-01-31 by mrbbking