olentangyfrc/InfiniteRecharge-2020 #14: Make sure all Motor Testing Controllers are in working order and finish Manual

NateRaccoon26, 2020-01-11:
I started revamping our Motor Testers last year including making the new CCM5D controllers.
We need to make sure all of our Motor Testers are working and if we have the time, improve them and if needed, fix them.

and one last little thing, I made Manual on how to use and more importantly fix these things last year but we never had time to print it out and make it into a book. I'd like to make time to do this.
LINK TO MANUAL:[https://docs.google.com/document/d/116YwsUeHxB8-QLLvmIuwFhfQNq-1ILW897hOt_1DNcw/edit?usp=sharing](https://docs.google.com/document/d/116YwsUeHxB8-QLLvmIuwFhfQNq-1ILW897hOt_1DNcw/edit?usp=sharing)

NateRaccoon26, 2020-01-11:
I also believe we didn't finish the box that contains the CCM5D Controllers @Eiim any chance you can get one of CAD people on this? I can talk to them tomorrow

NateRaccoon26, 2020-01-11:
If I had the ability or if someone sees this that has the ability, I'd like to assign, @mrbbking @sydneylewis22 @NateRaccoon26 @Eiim and whoever he wants working on the model

Eiim, 2020-01-11:
@NateRaccoon26 Assignments added, talk to the other Ethan tomorrow. He should have access to the right files.

mrbbking, 2020-01-12:
Guiding principles here: Electrical safety, ease-of-use, durability.

RockyRaccoon26, 2020-01-12:
Currently Waiting on CAD Team to finish the model and get us prints.

@ethandavis239 We will need at least 3 of them.

RockyRaccoon26, 2020-01-14:
Will be printing housings today at school

mrbbking, 2020-01-25:
Can you get an update where we are with these boxes, @RockyRaccoon26 ? Would it be better to just buy some project boxes and drill/mill the openings we need?

mrbbking, 2020-02-06:
dunno why I closed this before, sorry about that.

What can we learn from @mikagika & friends about making our own with an Arduino?

Remember our guiding principles are: Electrical safety, ease-of-use, durability.

mrbbking, 2020-02-06:
Ooh, hey look at this @mikagika : https://www.arduino.cc/en/Reference/ServoWriteMicroseconds

mikagika, 2020-02-07:
If that's truly setting the pulse width, then yeah that, seems pretty straight forward. The other variable, which is apparently sometimes important is the time between pulses. 

It would be pretty trivial to try it out though. Although Saturday, my priority is really to get zombie tested. But I can talk to the kids about possibly trying this on Sunday when I'm not there. They should be able to clone what we did for the rioTester and just change the output.

RockyRaccoon26, 2020-02-10:
As of now, planning on using an arduino compatible with a display, On/Off button, and a Pot.
We have the arduino code working, @mrbbking any chance you can upload yours or a pastebin of it here?

I believe we have 2 options as of now,
Use a Trinket M0 or a ItsyBitsy

If we go with a Trinket, We will be limited to a 4 digit segment display (that communicates over I2

if we go with an ItsyBitsy, we can go with a full 16x2 Character display

Issues with the Trinket, It works over I2C a communication protocol that at i dont have experience with, has only 5 pins which could be an issue

Issues with the Itsy Bitsy, Slightly bigger, a dollar more,

Im sure im missing things but this is what was on my mind

mikagika, 2020-02-10:
It looks to me like a Trinket M0 should work. I'm sure there are I2C libraries. Also, there are 16x2 displays that can be driven off from I2C. 
https://www.amazon.com/SunFounder-Serial-Module-Display-Arduino/dp/B019K5X53O
Or 
https://www.amazon.com/GeeekPi-Character-Backlight-Raspberry-Electrical/dp/B07S7PJYM6

That would probably be my preference (vs. the parallel connections we're currently using just because that's what we had available) if I was purchasing new components for a project.

mrbbking, 2020-02-11:
I picked up some things from MicroCenter today to make a prototype. I chose these because that's what they had and I want to get something working so people can use it and see how it performs. We can tweak from there once we have a working "something"

Itsy Bitsy
4-digit, 7-segment display (like an alarm clock) w/ I2C backpack
Crazy-tiny 128 x 32 OLED - about the size of the one in the Beak. Also I2C comms.

mrbbking, 2020-02-11:
Here's the code I was running at last meeting, @RockyRaccoon26 

```
// Started from Adafruit "Servo" example.
// Changed a bit.

#include <Servo.h> 
Servo servo;

int servoPin = 9;
int potPin   = 0;  
int speed    = 0;  // 0 - 100 percent, either direction
int dir      = 0;  // 1 = reverse, 0 = forward

// for updating the serial monitor every second during dev.
// replace with whatever the actual readout will be when built
// and omit the interval: update continuously.
unsigned long previousMillis = 0;
const long updateInterval = 1000;

void setup() 
{ 
  servo.attach(servoPin);
  Serial.begin(9600);
} 
 
void loop() 
{ 
  unsigned long currentMillis = millis();
  int reading = analogRead(potPin);           // analogRead returns 0 to 1023
  int ms = map(reading, 0, 1023, 1000, 2000); // change that to 1000 to 2000 us
  servo.writeMicroseconds(ms);

  if( ms >= 1475 && ms <= 1525 )              // per Spark Max manual
  {
    speed = 0;
  }
  if( ms < 1475 )                             // reverse, per Spark Max manual
  {
    speed = map( ms, 1000, 1475, 99, 0 );      // map ms to percentage
    dir = 1;
  }

  if( ms > 1525 )                             // forward, per Spark Max manual
  {
    speed = map(ms, 1525, 2000, 0, 99 );       // map ms to percentage
    dir = 0;
  }

 // for dev only, while using Serial monitor.
 // drop the "if" and replace the writing with whatever our actual display will be.
  if( currentMillis - previousMillis >= updateInterval)
  {
    previousMillis = currentMillis;
    Serial.print("\n\nDirection: ");
    if( dir ){
      Serial.println("Forward");
    } else {
      Serial.println("Reverse");
    }
    
    Serial.print("ms   : ");
    Serial.println(String(ms));
    Serial.print("Speed: ");
    Serial.println(String(speed));

  }
} 
```