olentangyfrc/InfiniteRecharge-2020 #16: Create Battery Check-in / Check-out sheets

mrbbking, 2020-01-11:
These sheets will be for logging battery data when one is taken and when one is returned so that we can track performance indicators all season long. This will help us identify the best performing batteries for competition and predict when any given battery may start to fail.

I anticipate this to be a clipboard hanging in the battery charging station with custom forms that prompt for two key values (internal resistance and state of charge) along with an indicator for "taking" or "returning" and the date & time.

This must not introduce too much friction: batteries must remain readily available.

RockyRaccoon26, 2020-01-11:
[How about something like this?](https://docs.google.com/document/d/1h4Q_OIjwNK9Gnk2AC5P9mKMU3y9X96fzDx_P1N16ACA/edit?usp=sharing)

mrbbking, 2020-01-12:
Good idea, content-wise, but we need something quick and easy to use in-the-moment or people will ignore them.

I'm pretty dead set on a paper sheet at the charging box for collecting the info, at least to start with. I have a spreadsheet I'm playing with for keeping track over time.

I'll bring something new for Sunday and we can look at that together.

RockyRaccoon26, 2020-01-12:
@mrbbking That was made Last night, I was thinking like a google form, We can add Reference Numbers to the batteries to make them faster to write (or maybe type) down.

If I have time, I'll draft one up tonight.

RockyRaccoon26, 2020-01-12:
[Heres a Form I quickly made, we could put a QR code that's easily scanned and completed](https://forms.gle/iH7LmX7E9wZKF5d27)
It just depends on what's easier, easier the better, This one is easier on us because now we don't enter the data ourselves and can be directly imported into a spreadsheet but honestly not sure which is easier for the rest of everyone else.

closed 2020-01-18 by mrbbking