olentangyfrc/InfiniteRecharge-2020 #20: General Housekeeping for Electrical

RockyRaccoon26, 2020-01-12:
### As we wait to get horribly busy in the following weeks we should finish some housekeeping tasks now.
- [ ] The Electrical Cabinet (The black one) Needs to be organized and if we have time new foam would be nice.

- [ ] We need to test a lot of our components, so we actually know which ones work and not waste time troubleshooting.

- [ ] Until we start soldering, ALL CAN wiring coming off of Components should be using Power Pole connectors and not Ferrule Connectors, In order to Connect to things that require Ferrule, There should be an "adapter" wire between them. This will save time and wire as the wires on things like the motor controllers once they are cut cannot be extended.

Everyone should be doing this where there is a lack of better things to do.

@mrbbking If you have anything to add (or tweak) that would be awesome.

mrbbking, 2020-01-18:
I would love to see a "wire termination kit" or two. A single box with powerpole and ferrule ends, crimpers, wire strippers, electrical tape, small flathead screwdriver. A single thing to grab when doing that kind of work.

Let's also get rid of anything that's broken (e.g. wire cutters with excessive nicks on the blades)

RockyRaccoon26, 2020-02-11:
Closing as we begin the busy part of the season for electrical, also the main items on this list have been completed. We can reopen later if needed

closed 2020-02-11 by RockyRaccoon26