olentangyfrc/InfiniteRecharge-2020 #51: Code/Build roboRIO hardware tester and test "bad" roboRIO

mikagika, 2020-01-19:
I believe I've researched and done enough POC to show that we can effectively test the "bad' rio via an arduino. Should be able to test at least DIO, AIO, and PWM ports. Have basic idea of code to do so on both Arduino and RIO side. Want to recruit some kids and build this out, and then eventually make the unit into a permanent test tool. May also be useful for discussing basics of signalling and how things actually talk to each other.

RockyRaccoon26, 2020-01-21:
Once you guys get the code section all set up, electrical can definitely help build this (or at the very least, I would :) )

mikagika, 2020-01-21:
Thanks! Don't expect it to be ready "tomorrow" though. It will take us a
while to sort out all the details of what we want it to do.