olentangyfrc/InfiniteRecharge-2020 #70: Test endurance of batteries on swerve chassis

mrbbking, 2020-01-25:
I would like to get two full runs of the swerve drive chassis from a fully-charged battery until brownout to see what the draw is with the swerve drive and how long a good battery can supply it.

This should require no more than 10 minutes total drive time.

I'd like to be able to see the power-related graphs & data on the drive station.

I'd like to get an uncut video of the whole drive time so we can correlate what the graphs tell us with what the bot was doing at the time.

@olentangyfrcowner @worthp @topfuel388 @mikagika @JenniferKing4611

JenniferKing4611, 2020-01-25:
For the video, please pull  Tanish Ravuri for this.  I'd recommend explaining exactly what you are trying to get out of the video so that he knows what to focus on.

mrbbking, 2020-02-21:
I'd like this to happen at every opportunity while we're driving from here on out - take advantage of the driving that'll already be happening.

Get logs off the drive station so we can look at them on a different computer & not tie that one up. Sam King knows where to find the logs.