olentangyfrc/InfiniteRecharge-2020 #89: FYI: Specs & sample code for beam break sensors

mrbbking, 2020-02-08:
The sensors are item 2168 from Adafruit (http://adafru.it/2168).
They  are "open collector transistor output" which means they need a pull-up resistor, which the RoboRIO has built in according to the [WpiLib DIO docs](https://docs.wpilib.org/en/latest/docs/hardware/sensors/digital-inputs-hardware.html). So they should work as-wired.

Testing with an Arduino at home and with the receiver and transmitter 12" apart on my desktop, I slowly lifted the receiver straight off the desktop, keeping it aimed horizontally. The beam did not break until it was 3" up (about 15 degrees, if my trig is right).

The Arduino code I used is what Adafruit publishes at https://learn.adafruit.com/ir-breakbeam-sensors/arduino

mrbbking, 2020-02-08:
I'm not asking for any action or resolution, here. Just sharing what I learned & hoping to maybe save the coders some time & trouble.

mikagika, 2020-02-08:
Cool, I'm thinking that should work pretty well.