olentangyfrc/InfiniteRecharge-2020 #100: DuPont (i.e. "tiny") Connectors: Good Crimp example

mrbbking, 2020-02-10:
The photos on this page show good, solid crimps with what appears to be the exact connectors in the kit that @mikagika is so kindly letting us use. Please look at them to see what a successful crimp looks like (scroll halfway down).

Please leave a comment on this issue to let me know you've seen it & I'll close the issue once you've all done that.

I'll also put a reference in the binder we're making so it's available in-the-moment next time we need to make these.

https://www.pololu.com/category/71/wires-with-pre-crimped-terminals

Direct image link (if this works): https://a.pololu-files.com/picture/0J2541.1200.jpg

RockyRaccoon26, 2020-02-10:
comment

isabellebubnick, 2020-02-10:
I looked at the pictures